HKUST-Aerial-Robotics/Fast-Planner #151: I click on the 2D Nav Goal on the Rviz, but it didn;t work

lijoe123, 2022-10-20:
When i run the `roslaunch plan_manage kino_replan.launch`, it occur this error:
[image]
By the way ,when i run this command `roslaunch plan_manage kino_replan.launch`, it can show the map and the plane, but i click the 2D Nav Goal, it didn't work.

I used the fast planner ubuntu 20.04 version.
Thank you a lot!

Hao-Starrr, 2022-11-06:
Same problem. I uninstalled the ros-noetic-nlopt and install it from official github. It still doesn't work.

lijoe123, 2022-11-07:
I had fixed this problem by [](https://blog.csdn.net/asd22222984565/article/details/126779002#comments_23784587)
this part:
[image]

DylanWang1002, 2024-01-29:
> I had fixed this problem by (https://blog.csdn.net/asd22222984565/article/details/126779002#comments_23784587) this part: [image]

thx a lot

tl123987, 2024-04-24:
good

1091205886, 2024-09-02:
> > I had fixed this problem by (https://blog.csdn.net/asd22222984565/article/details/126779002#comments_23784587) this part: [image]
> 
> thx a lot

**I fallow that, but still appear error after I clicking on the Rviz, like: **
[image]