HKUST-Aerial-Robotics/Fast-Planner #156: About the Perception-aware Replanning

KingStrong-bit, 2022-12-13:
Hi,

I have read the paper **PAPTRO** and it's perfect! 
I want to know  when will the Perception-aware Replanning code be released?

Thanks

FPSychotic, 2022-12-13:
> Hi,
> 
> I have read the paper **PAPTRO** and it's perfect! I want to know when will the Perception-aware Replanning code be released?
> 
> Thanks

Any link to such paper? Thanks, you wake up my curiosity XD

KingStrong-bit, 2022-12-13:
> > Hi,
> > I have read the paper **PAPTRO** and it's perfect! I want to know when will the Perception-aware Replanning code be released?
> > Thanks
> 
> Any link to such paper? Thanks, you wake up my curiosity XD

Sorry, I wrote the title wrong, it should be ‘RAPTOR: Robust and Perception-aware Trajectory Replanning for Quadrotor Fast Flight’