HKUST-Aerial-Robotics/Fast-Planner #175: segfault in fast_planner_node on jetson orin Linux 35.5.0

lida2003, 2024-06-05:
Latest code on github and segfault when I click in simulator.

Any idea or suggestion?

```
[FSM]: state: INIT
wait for goal.
[FSM]: state: INIT
wait for goal.
[FSM]: state: INIT
wait for goal.
[FSM]: state: INIT
wait for goal.
[FSM]: state: INIT
wait for goal.
Triggered!
[FSM]: from INIT to WAIT_TARGET
[FSM]: from WAIT_TARGET to GEN_NEW_TRAJ
[kino replan]: -----------------------
start:         -19           0 -2.1684e-19, 0 0 0, 0 0 0
goal:-8.71761  1.11666        1, 0 0 0
Stack trace (most recent call last):
#13   Object "/home/daniel/HKUST/devel/lib/plan_manage/fast_planner_node", at 0xaaaad834a41b, in 
#12   Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffff967afe0f, in __libc_start_main
#11   Object "/home/daniel/HKUST/devel/lib/plan_manage/fast_planner_node", at 0xaaaad8347f7f, in main
#10   Object "/opt/ros/noetic/lib/libroscpp.so", at 0xffff97178a07, in ros::spin()
#9    Object "/opt/ros/noetic/lib/libroscpp.so", at 0xffff971906c7, in ros::SingleThreadedSpinner::spin(ros::CallbackQueue*)
#8    Object "/opt/ros/noetic/lib/libroscpp.so", at 0xffff9713a193, in ros::CallbackQueue::callAvailable(ros::WallDuration)
#7    Object "/opt/ros/noetic/lib/libroscpp.so", at 0xffff97138a6b, in ros::CallbackQueue::callOneCB(ros::CallbackQueue::TLS*)
#6    Object "/opt/ros/noetic/lib/libroscpp.so", at 0xffff97114bbf, in ros::TimerManager<ros::Time, ros::Duration, ros::TimerEvent>::TimerQueueCallback::call()
#5    Object "/home/daniel/HKUST/devel/lib/plan_manage/fast_planner_node", at 0xaaaad8350b8f, in fast_planner::KinoReplanFSM::execFSMCallback(ros::TimerEvent const&)
#4    Object "/home/daniel/HKUST/devel/lib/plan_manage/fast_planner_node", at 0xaaaad835045b, in fast_planner::KinoReplanFSM::callKinodynamicReplan()
#3    Object "/home/daniel/HKUST/devel/lib/plan_manage/fast_planner_node", at 0xaaaad8367723, in fast_planner::FastPlannerManager::kinodynamicReplan(Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>)
#2    Object "/home/daniel/HKUST/devel/lib/libpath_searching.so", at 0xffff9734bce3, in fast_planner::KinodynamicAstar::search(Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, Eigen::Matrix<double, 3, 1, 0, 3, 1>, bool, bool, double)
#1    Object "/home/daniel/HKUST/devel/lib/libpath_searching.so", at 0xffff973453db, in 
#0    Object "/lib/aarch64-linux-gnu/libc.so.6", at 0xffff96815690, in 
Segmentation fault (Address not mapped to object [0xa])
[fast_planner_node-1] process has died [pid 9919, exit code -11, cmd /home/daniel/HKUST/devel/lib/plan_manage/fast_planner_node /odom_world:=/state_ukf/odom /sdf_map/odom:=/state_ukf/odom /sdf_map/cloud:=/pcl_render_node/cloud /sdf_map/pose:=/pcl_render_node/camera_pose /sdf_map/depth:=/pcl_render_node/depth __name:=fast_planner_node __log:=/home/daniel/.ros/log/b044125c-22de-11ef-9440-200b7460c5b7/fast_planner_node-1.log].
log file: /home/daniel/.ros/log/b044125c-22de-11ef-9440-200b7460c5b7/fast_planner_node-1*.log
^C[pcl_render_node-9] killing on exit
[odom_visualization-8] killing on exit
[so3_disturbance_generator-7] killing on exit
[so3_control-6] killing on exit
[quadrotor_simulator_so3-5] killing on exit
[random_forest-4] killing on exit
[waypoint_generator-3] killing on exit
[traj_server-2] killing on exit
terminate called after throwing an instance of 'boost::wrapexcept<boost::lock_error>'
  what():  boost: mutex lock failed in pthread_mutex_lock: Invalid argument
shutting down processing monitor...
... shutting down processing monitor complete
done
```

detailed log: [b044125c-22de-11ef-9440-200b7460c5b7.zip](https://github.com/user-attachments/files/15572195/b044125c-22de-11ef-9440-200b7460c5b7.zip)

IngaVS, 2024-09-11:
timeToIndex function doesn't return the value.  
`int KinodynamicAstar::timeToIndex(double time)
{
  int idx = floor((time - time_origin_) * inv_time_resolution_);
  return idx;
}`

lida2003, 2024-09-11:
@IngaVS Thanks, great job. Can you make a PR?